RethinkRobotics/baxter_simulator #78: joint_state_controller run dependency

kmhallen, 2016-01-13:
After installing ```ros-indigo-baxter-simulator``` with apt-get and running 
```roslaunch baxter_gazebo baxter_world.launch```, the following error came up.

```
[ERROR] [1452721279.691889491, 0.542000000]: Could not load controller 'joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.
[ERROR] [1452721279.691927138, 0.542000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
```

joint_state_controller or its meta-package (ros_controllers) should be added as a run dependency of either baxter_sim_controllers or baxter_gazebo.

rethink-imcmahon, 2016-01-13:
Good find. [baxter_sim_hardware/baxter_sdk_control.launch](https://github.com/RethinkRobotics/baxter_simulator/blob/49a0ab2e349f248ff9066883565629cc8be7c22e/baxter_sim_hardware/launch/baxter_sdk_control.launch#L22-L23) file is the offender:
```
  <!-- load the default controllers -->
  <node name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
        output="screen" ns="/robot" args="--shutdown-timeout -1 joint_state_controller" />
```

davetcoleman, 2017-02-14:
This PR would fix it: https://github.com/RethinkRobotics/baxter_simulator/pull/84

IanTheEngineer, 2017-02-20:
Resolved by #84

closed 2017-02-20 by IanTheEngineer