RethinkRobotics/baxter_simulator #82: Error when executing roslaunch baxter_gazebo baxter_world.launch

KahnShi, 2016-03-26:
Dear all,

My system is Ubuntu 12.04 and I am using Hydro.

I did as this tutoial:
https://github.com/RethinkRobotics/sdk-docs/wiki/Baxter-simulator

When I try to `roslaunch baxter_gazebo baxter_world.launch`
the ouput is:

> Invalid tag: Cannot load command parameter [robot_description]: command [../ros_ws/src/xacro/xacro.py --inorder ../ros_ws/src/baxter_common/baxter_description/urdf/baxter.urdf.xacro gazebo:=true] returned with code [2].

Then I tried in this way
`kahn@kahn:~/ros/ws_baxter$ /home/kahn/ros/ws_baxter/src/xacro/xacro.py --inorder` 
the ouput is

> /home/kahn/ros/ws_baxter/src/baxter_common/baxter_description/urdf/baxter.urdf.xacro gazebo:=true
> option --inorder not recognized
> Usage: xacro.py [-o <output>] <input>
>        xacro.py --deps       Prints dependencies
>        xacro.py --includes   Only evalutes includes

Then I directly tried this
`kahn@kahn:~/ros/ws_baxter$ /home/kahn/ros/ws_baxter/src/xacro/xacro.py ` 
However, there are still mistakes..

> /home/kahn/ros/ws_baxter/src/baxter_common/baxter_description/urdf/baxter.urdf.xacro gazebo:=true
> Traceback (most recent call last):
>   File "/home/kahn/ros/ws_baxter/src/xacro/xacro.py", line 60, in <module>
>     xacro.main()
>   File "<string>", line 657, in main
>   File "<string>", line 245, in process_includes
>   File "<string>", line 474, in eval_text
>   File "<string>", line 461, in handle_expr
>   File "<string>", line 435, in eval_expr
>   File "<string>", line 409, in eval_term
>   File "<string>", line 389, in eval_factor
>   File "<string>", line 367, in eval_lit
> xacro.XacroException: Property wasn't defined: u'l_finger'
> 

I think my problem is totally same with this guy, but his use Indigo
http://answers.gazebosim.org/question/12765/error-when-executing-roslaunch-baxter_gazebo-baxter_worldlaunch/

Could anyone give me some advice?

Thx in advance!

IanTheEngineer, 2016-03-26:
Hi there. It seems your Xacro package is out of date. Try to upgrade it to the most recent:

`$ sudo apt-get install ros-indigo-xacro`

You may need to upgrade all of your packages:

`$ sudo apt-get update && sudo apt-get dist-upgrade`

IanTheEngineer, 2016-03-26:
Ah. ROS Hydro and 12.04. The RSDK 1.2 release is explicitly for Ubuntu 14.04 and ROS Indigo. You can either:
- Install RSDK 1.1
- Install Xacro from ROS Indigo source, and try to make it work
- Upgrade your machine / ROS

KahnShi, 2016-03-27:
Thank you for your help. Now it worked.

And I wish to apply some advice for future users with Hydro version.

It works when you download baxter, baxter_common, baxter_examples, baxter_interface, you just choose branch of release-1.1.1.

But for baxter_simulator, they don't have 1.1.1, so I choose to use 0.9.1.

All these files could be downloaded in https://github.com/RethinkRobotics

Then enjoy it.

closed 2016-03-27 by KahnShi