RethinkRobotics/baxter_simulator #85: Missing SDK version 1.1.1 for the simulator

m-talha, 2016-04-14:
Hi

I'm trying to run the Baxter in simulation but when I run my script it gives an error:

Baxter SDK version 2.1.2 not the same as local SDK version 1.1.1.

How can this be overcome? I need to use v1.1.1 as that is on the real Baxter. 

Thanks

IanTheEngineer, 2016-04-14:
Ah, the simulator tagging was a bit confusing. You need version `0.9.1`, which corresponded to `1.1.1`. Just checkout the release branch for `1.1.1` and you should be good to go. Just keep in mind, grippers don't exist in simulation until `1.2.0`

m-talha, 2016-04-15:
Ok but now I get no movement in Gazebo and I get an error when doing position control:

```
[ERROR] [1460733123.160097390, 117.938000000]: Could not stop controller 'left_joint_velocity_controller' since it is not running
[ERROR] [1460733123.247882722, 117.952000000]: Failed to switch controllers
```

IanTheEngineer, 2016-04-15:
Make sure you have all of the dependencies installed on your workstation. See the installation tutorial for details:
http://sdk.rethinkrobotics.com/wiki/Simulator_Installation

m-talha, 2016-04-19:
Following the simulator installation tutorial on a different computer and running catkin_make to install the baxter simulator fails and results in multiple errors:

`/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp: In member function ‘bool baxter_sim_io::QNode::init()’:
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:117:23: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::left_arm_nav.button_names.push_back("ok");
                       ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:118:23: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::left_arm_nav.button_names.push_back("back");
                       ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:119:23: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::left_arm_nav.button_names.push_back("show");
                       ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:123:23: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::left_arm_nav.light_names.push_back("inner");
                       ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:124:23: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::left_arm_nav.light_names.push_back("outer");
                       ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:129:24: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::right_arm_nav.button_names.push_back("ok");
                        ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:130:24: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::right_arm_nav.button_names.push_back("back");
                        ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:131:24: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::right_arm_nav.button_names.push_back("show");
                        ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:135:24: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::right_arm_nav.light_names.push_back("inner");
                        ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:136:24: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::right_arm_nav.light_names.push_back("outer");
                        ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:141:28: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::left_shoulder_nav.button_names.push_back("ok");
                            ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:142:28: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::left_shoulder_nav.button_names.push_back("back");
                            ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:143:28: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::left_shoulder_nav.button_names.push_back("show");
                            ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:147:28: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::left_shoulder_nav.light_names.push_back("inner");
                            ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:148:28: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::left_shoulder_nav.light_names.push_back("outer");
                            ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:153:29: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::right_shoulder_nav.button_names.push_back("ok");
                             ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:154:29: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::right_shoulder_nav.button_names.push_back("back");
                             ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:155:29: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘button_names’
   QNode::right_shoulder_nav.button_names.push_back("show");
                             ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:159:29: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::right_shoulder_nav.light_names.push_back("inner");
                             ^
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_io/src/qnode.cpp:160:29: error: ‘baxter_core_msgs::NavigatorState’ has no member named ‘light_names’
   QNode::right_shoulder_nav.light_names.push_back("outer");
                             ^
make[2]: *** [baxter_simulator/baxter_sim_io/CMakeFiles/baxter_sim_io.dir/src/qnode.cpp.o] Error 1
make[1]: *** [baxter_simulator/baxter_sim_io/CMakeFiles/baxter_sim_io.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_hardware/src/baxter_emulator.cpp: In member function ‘bool baxter_en::baxter_emulator::init()’:
[... cut ...]
/home/baxter/ros_ws/src/baxter_simulator/baxter_sim_hardware/src/baxter_emulator.cpp:516:18: error: ‘baxter_core_msgs::HeadState’ has no member named ‘isTurning’
         head_msg.isTurning = false;
                  ^
make[2]: *** [baxter_simulator/baxter_sim_hardware/CMakeFiles/baxter_emulator.dir/src/baxter_emulator.cpp.o] Error 1
make[1]: *** [baxter_simulator/baxter_sim_hardware/CMakeFiles/baxter_emulator.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`
Baxter SDK is version 1.1.1 and simulator is 0.9.1.

IanTheEngineer, 2016-04-19:
It looks like your `baxter_common` repo is on `v1.2.0`  aka` master`, which would break the build. Just update your `rosinstall` as follows to switch over ALL of your baxter repositories to `v1.1.1`:
```
$ cd ~/ros_ws/src
$ wstool init .
$ wstool merge https://raw.githubusercontent.com/RethinkRobotics/baxter_simulator/release-0.9.1/baxter_simulator.rosinstall
$ wstool update
$ cd ~/ros_ws
$ catkin_make
```

closed 2016-04-22 by IanTheEngineer