RethinkRobotics/baxter_simulator #87: Build issue with ros kinetic and ubuntu 16.04

arpit15, 2016-10-04:
[error](https://gist.github.com/arpit15/cdd33e11fa3b7fd4c0f2f04ea598efb9) 
I am facing above errors when running catkin_make with ros kinetic and ubuntu 16.04. Please let me know what are the steps required to be done to compile this package properly.

IanTheEngineer, 2016-10-04:
Hi @arpit15. `baxter_simulator` does not support ROS Kinetic and Ubuntu 16.04. We recommend using ROS Indigo and Ubuntu 14.04. If you do manage to get the `baxter_simulator` to work in Kinetic, please open a pull request / ticket detailing the changes you made. Thank you.

closed 2016-10-04 by IanTheEngineer

arpit15, 2016-10-04:
I made small changes to baxter_simulator/baxter_sim_io/include/baxter_sim_io/qnode.hpp due to known [issue](https://github.com/ros-planning/moveit_ros/issues/653) with Qt4.8.7 and boost 1.58 .I was able to compile and run the basic example `rosrun baxter_gazebo baxter_world.launch`. Let me know if I should submit a PR with this change. I will try out other examples as well in the meantime.