RethinkRobotics/baxter_simulator #89: Unable to install baxter simulator. Ros: Indigo OS: Ubuntu 14:04

mjm522, 2016-10-24:
I tried installing baxter simulator several times, but keep getting this error always. I tried deleting the devel folder and started from the beginning, but with no progress. Can anybody please tell me what I doing wrong? 

`/usr/local/include/urdf_model/types.h:62:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(JointMimic);
 ^
/usr/local/include/urdf_model/types.h:62:1: error: ‘weak_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(JointMimic);
 ^
/usr/local/include/urdf_model/types.h:63:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(JointSafety);
 ^
/usr/local/include/urdf_model/types.h:63:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(JointSafety);
 ^
/usr/local/include/urdf_model/types.h:63:1: error: ‘weak_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(JointSafety);
 ^
/usr/local/include/urdf_model/types.h:64:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Link);
 ^
/usr/local/include/urdf_model/types.h:64:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Link);
 ^
/usr/local/include/urdf_model/types.h:64:1: error: ‘weak_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Link);
 ^
/usr/local/include/urdf_model/types.h:65:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Material);
 ^
/usr/local/include/urdf_model/types.h:65:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Material);
 ^
/usr/local/include/urdf_model/types.h:65:1: error: ‘weak_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Material);
 ^
/usr/local/include/urdf_model/types.h:66:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Mesh);
 ^
/usr/local/include/urdf_model/types.h:66:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Mesh);
 ^
/usr/local/include/urdf_model/types.h:66:1: error: ‘weak_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Mesh);
 ^
/usr/local/include/urdf_model/types.h:67:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Sphere);
 ^
/usr/local/include/urdf_model/types.h:67:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Sphere);
 ^
/usr/local/include/urdf_model/types.h:67:1: error: ‘weak_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Sphere);
 ^
/usr/local/include/urdf_model/types.h:68:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Visual);
 ^
/usr/local/include/urdf_model/types.h:68:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Visual);
 ^
/usr/local/include/urdf_model/types.h:68:1: error: ‘weak_ptr’ in namespace ‘std’ does not name a type
 URDF_TYPEDEF_CLASS_POINTER(Visual);
 ^
/usr/local/include/urdf_model/types.h:72:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 std::shared_ptr<T> const_pointer_cast(std::shared_ptr<U> const & r)
 ^
/usr/local/include/urdf_model/types.h:78:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 std::shared_ptr<T> dynamic_pointer_cast(std::shared_ptr<U> const & r)
 ^
/usr/local/include/urdf_model/types.h:84:1: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 std::shared_ptr<T> static_pointer_cast(std::shared_ptr<U> const & r)
[... 338 lines cut ...]
/usr/local/include/urdf_model/model.h:163:78: error: request for member ‘begin’ in ‘((urdf::ModelInterface*)this)->urdf::ModelInterface::links_’, which is of non-class type ‘int’
     for (std::map<std::string, LinkSharedPtr>::const_iterator l=this->links_.begin(); l!=this->links_.end(); l++)
                                                                              ^
/usr/local/include/urdf_model/model.h:163:103: error: request for member ‘end’ in ‘((urdf::ModelInterface*)this)->urdf::ModelInterface::links_’, which is of non-class type ‘int’
     for (std::map<std::string, LinkSharedPtr>::const_iterator l=this->links_.begin(); l!=this->links_.end(); l++)
                                                                                                       ^
/usr/local/include/urdf_model/model.h:163:108: error: expected ‘)’ before ‘;’ token
     for (std::map<std::string, LinkSharedPtr>::const_iterator l=this->links_.begin(); l!=this->links_.end(); l++)
                                                                                                            ^
/usr/local/include/urdf_model/model.h:163:110: error: ‘l’ was not declared in this scope
     for (std::map<std::string, LinkSharedPtr>::const_iterator l=this->links_.begin(); l!=this->links_.end(); l++)
                                                                                                              ^
/usr/local/include/urdf_model/model.h:163:113: error: expected ‘;’ before ‘)’ token
     for (std::map<std::string, LinkSharedPtr>::const_iterator l=this->links_.begin(); l!=this->links_.end(); l++)
                                                                                                                 ^
/usr/local/include/urdf_model/model.h:184:3: error: expected ‘}’ at end of input
   }
   ^
Linking CXX executable /home/mjm/baxter_ws/devel/lib/baxter_sim_io/baxter_sim_io
[ 97%] Built target baxter_sim_io
make[2]: *** [baxter_simulator/baxter_gazebo/CMakeFiles/baxter_gazebo_ros_control.dir/src/baxter_gazebo_ros_control_plugin.cpp.o] Error 1
make[1]: *** [baxter_simulator/baxter_gazebo/CMakeFiles/baxter_gazebo_ros_control.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

mjm522, 2016-10-24:
What I did: I am interested in simulator version 0.9.1

```
$cd baxter_ws/src/
$git clone https://github.com/RethinkRobotics/baxter_simulator.git
$cd baxter_simulator
$git checkout release-0.9.1
$cd ../../
$source /opt/ros/indigo/setup.bash
$ catkin_make
$ catkin_make install
```

mjm522, 2016-10-24:
I also tried: the instructions from webpage; with no result.

```
$ cd ~/ros_ws/src
$ wstool init .
$ wstool merge https://raw.githubusercontent.com/RethinkRobotics/baxter_simulator/master/baxter_simulator.rosinstall
$ wstool update
$source /opt/ros/indigo/setup.bash
$ catkin_make
$ catkin_make install
```

knorth55, 2016-10-24:
have you tried `rosdep install`?
if not, try it after wstool update.

```
cd ~/ros_ws/src
wstool init .
wstool merge https://raw.githubusercontent.com/RethinkRobotics/baxter_simulator/master/baxter_simulator.rosinstall
wstool update
rosdep install --from-path . --ignore-src -y -r
source /opt/ros/indigo/setup.bash
catkin build
```

mjm522, 2016-10-24:
Thanks for the reply. I get all the same errors are before plus following new errors:

```
Failed     << baxter_sim_kinematics:make             [ Exited with code 2 ]    
Failed    <<< baxter_sim_kinematics                  [ 7.7 seconds ]           
Finished  <<< baxter_sim_io                          [ 8.8 seconds ]           
[build] Summary: 9 of 14 packages succeeded.                                   
[build]   Ignored:   3 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 3 packages were abandoned.                                
[build]   Failed:    2 packages failed.                                        
[build] Runtime: 13.9 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```
I am able to isntall packages baxter_common, baxter_interface, baxter_pykdl, baxter, baxter_examples.

But the baxter_simulator does get installed. The errors seem like some C++ compiler fault, like it is not able to resolve some data types. But I don't know where and how to edit to get past this error.

knorth55, 2016-10-24:
@mjm522 my compiler version is below, and I can compile the package.
try to upgrade your compiler.
```
$ g++ --version
g++ (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

mjm522, 2016-10-24:
mine is 

```
$ g++ --version
g++ (Ubuntu 4.8.5-2ubuntu1~14.04.1) 4.8.5
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
It is not that issue. I am still trying to figure out.

knorth55, 2016-10-24:
@mjm522 
according to these FAQ, it seems urdf header file problem.
can you check liburdfdom-headers-dev version?

http://answers.ros.org/question/243756/ros-indigo-install-on-rpi3-urdf-compile-errors/
https://github.com/ros/urdfdom_headers/releases/tag/1.0.0
```
$ dpkg --status liburdfdom-headers-dev
Package: liburdfdom-headers-dev
Status: install ok installed
Priority: extra
Section: libdevel
Installed-Size: 115
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Multi-Arch: foreign
Source: urdfdom-headers
Version: 0.2.3+dfsg-1
Depends: pkg-config
Description: URDF DOM - header files
 The URDF (U-Robot Description Format) library provides core data
 structures and a simple XML parsers for populating the class data
 structures from an URDF file.
 .
 This package contains the headers files.
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>

```

knorth55, 2016-10-24:
@mjm522 
i updated liburdfdom-headers-dev to 1.0.0 and i got same error.
I think it is urdfdom-header problem.
my solution is downgrade urdfdom-headers to 0.4.2 and report this issue https://github.com/ros/urdfdom_headers/issues.
i use really old urdfdom-headers, but it doesn't matter for my baxter simulation.

mjm522, 2016-10-24:
Yes, exactly, that fixes the problem. Thanks a lot.  :+1:

closed 2016-10-24 by mjm522

IanTheEngineer, 2016-10-24:
Thanks for your help debugging @knorth55 . Looks like this is a regression in urdfdom-headers

knorth55, 2016-10-24:
@mjm522 I'm glad to help you :)
@IanTheEngineer you're welcome