RethinkRobotics/baxter_simulator #94: Kinetic & Gazebo7: Video streaming fails

IanTheEngineer, 2017-02-03:
Baxter's camera streaming fails using Gazebo7:
```
[ WARN] [1486153355.581155367]: GazeboRosVideo plugin missing <robotNamespace>, defaults to "".
[ WARN] [1486153355.581207155]: GazeboRosVideo Plugin (ns = ) missing <topicName>, defaults to "image_raw".
[ WARN] [1486153355.581218927]: GazeboRosVideo Plugin (ns = ) missing <height>, defaults to 240.
[ WARN] [1486153355.581227121]: GazeboRosVideo Plugin (ns = ) missing <width>, defaults to 320
```

davetcoleman, 2017-02-14:
Camera streaming is working for me using the following three test commands:

```
rosrun image_view image_view image:=/cameras/head_camera/image
rosrun image_view image_view image:=/cameras/right_hand_camera/image
rosrun image_view image_view image:=/cameras/left_hand_camera/image
```

And your related [PR](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/534) that was merged in. 

Though I am also seeing those warnings in the console. You mentioned having more patches, do you have a fix for this already or should I create one?

davetcoleman, 2017-02-16:
This bug has already been reported on [Gazebo issue tracker](https://bitbucket.org/osrf/gazebo/issues/2103/visualplugins-not-registering-sdf-elements)

davetcoleman, 2017-02-16:
Seems like an issue with the SDF parser for URDFs, I spent considerable time on this today but have not solved it

IanTheEngineer, 2017-02-17:
Ah, I'm sorry Dave, I think I did get to the root of this last week, but this week I've had my hands full doing a bunch of infrastructure upgrades. I'll follow up on the two issues Camera streaming and Image displaying using `gazebo_ros_pkgs`. The camera issue seems to have to do with namespacing, and the Image display issue seems to be caused by Gazebo7 injecting extra SDF tags when parsing the Video plugin from URDF. I have patches for both, but they'll need OSRF feedback.

davetcoleman, 2017-02-17:
Oops my last two comments were in regards to the image display, and should have been reported in a different issue (https://github.com/RethinkRobotics/baxter_simulator/issues/95).

The problem with the camera streaming seems fixed, and the warnings only needs this PR merged: https://github.com/RethinkRobotics/baxter_common/pull/86