RethinkRobotics/baxter_simulator #95: Kinetic & Gazebo7: No head display

IanTheEngineer, 2017-02-03:
The head display image does not show up in Gazebo7:
[image]

davetcoleman, 2017-02-14:
I have confirmed this issue. Have you begun a fix for this yet?

davetcoleman, 2017-02-17:
This bug has already been reported on [Gazebo issue tracker](https://bitbucket.org/osrf/gazebo/issues/2103/visualplugins-not-registering-sdf-elements), but there is not a good solution for it. 

I agree with @IanTheEngineer that:

> Image display issue seems to be caused by Gazebo7 injecting extra SDF tags when parsing the Video plugin from URDF.

davetcoleman, 2017-02-19:
I have further tracked down and reported the bug [here](https://bitbucket.org/osrf/gazebo/issues/2202/gazebo7-visualplugins-are-deprecated)

davetcoleman, 2017-02-19:
Screen is working for me now with [this PR](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/546), but SDF warnings persist