RethinkRobotics/baxter_simulator #112: Gazebo start error

ilaplace, 2017-03-27:
Hello there, whenever I start baxter gazebo I'm getting this error although simulation the starts:
```
[ERROR] [1490634846.719733201]: Could not find parameter robot_base_description on parameter server
[ERROR] [1490634846.725947716]: RobotURDF: Parameter /robot_base_description could not be found/read
[ERROR] [1490634846.725990169]: robot_state_publisher:  failed to initialize!
```

Any idea what’s this about?

IanTheEngineer, 2017-03-27:
Have you cloned the [`mutable_robot_state_publisher`](https://github.com/RethinkRobotics/mutable_robot_state_publisher) into your workspace? I would recommend against using it in its current form, and just use the standard `robot_state_publisher` provided by ROS. Remove it from your workspace, then remove the `build/` and `devel/` directories, and rebuild:
```
$ source /opt/ros/indigo/setup.bash
$ catkin_make
$ ./baxter.sh sim
```

ilaplace, 2017-03-28:
You are right  I've downloaded mutable_state_publisher in order to replace Baxter's gripper with Allegro Hand. Nonetheless your advice didn't work for me I'm still getting the same error. :sob:

IanTheEngineer, 2017-03-28:
Alrighty - let's move this discussion back to the BRR-forum. The issue at hand is likely just how your workspace is set up. Please try creating a fresh workspace using this tutorial:
http://sdk.rethinkrobotics.com/wiki/Simulator_Installation

closed 2017-03-28 by IanTheEngineer