RethinkRobotics/baxter_simulator #122: Baxter_Gazebo: can't get current joint states

pipothy, 2018-04-17:
Hi,

I have an issue with running any of the code to move the robot. I get the below error when I run the joint position waypoints example:

$ rosrun baxter_examples joint_position_waypoints.py -l right Initializing node...  Traceback (most recent call last):   File "/home/pipothy/ros_ws/src/baxter_examples/scripts/joint_position_waypoints.py", line 191, in <module>
    main()   File "/home/pipothy/ros_ws/src/baxter_examples/scripts/joint_position_waypoints.py", line 181, in main
    waypoints = Waypoints(args.limb, args.speed, args.accuracy)   File "/home/pipothy/ros_ws/src/baxter_examples/scripts/joint_position_waypoints.py", line 45, in __init__
    self._limb = baxter_interface.Limb(self._arm)   File "/home/pipothy/ros_ws/src/baxter_interface/src/baxter_interface/limb.py", line 121, in __init__
    timeout_msg=err_msg)   File "/home/pipothy/ros_ws/src/baxter_interface/src/baxter_dataflow/wait_for.py", line 55, in wait_for
    raise OSError(errno.ETIMEDOUT, timeout_msg) OSError: [Errno 110] Right limb init failed to get current joint_states from robot/joint_states

I already had ROS installed, but other than that, I followed the installation instructions at the link below line by line. I used the commands for kinetic and stopped after launching the baxter world: http://sdk.rethinkrobotics.com/wiki/Simulator_Installation 
I then enabled the robot and ran the command to start the example: http://sdk.rethinkrobotics.com/wiki/Joint_Position_Waypoints_Example 
This produced the above error.

Am I being silly and have forgot to do something, or is this a genuine error? Apologies if it is me being silly, I can't seem to find anything in the rest of the issues.

knorth55, 2018-04-18:
I think you got the error in your first trial.
That is because gazebo download models in the first time and fail to launch baxter gazebo plugins correctly.
Can you try to relaunch again? and check `rostopic hz /robot/joint_states`?

pipothy, 2018-04-18:
Hi,

In short, yep, you're correct! Launching the world once, it failed in the usual way. But then, closing that down and relaunching the world resulted in it working.

Sorry for the trivial issue, but thanks so much for your help!

sxwgit, 2019-03-18:
Hello, I meet the same error.
I relaunch the world ,just like 
```
./baxter.sh sim
roslaunch baxter_gazebo baxter_world.launch
ctrl+z
roslaunch baxter_gazebo baxter_world.launch

```
I get the info 

> [ INFO] [1552887561.104179646, 21.821000000]: Simulator is loaded and started successfully

But it didn't solve the problem, and the same error still occurred.
Would you mind tell me how you solved this problem?
Best regards.

knorth55, 2019-03-18:
@sxwgit 
Please do not use `Ctrl + Z`.
First launch should take a long time, so please wait.
After first launching, you need to kill all gazebo process by `Ctrl + C`, and launch the same file again.

sxwgit, 2019-03-18:
@knorth55 
Thank you very much for your kind help.
I follow your method: 
```
./baxter.sh sim
roslaunch baxter_gazebo baxter_world.launch
# waited about 5 mins
ctrl + c
roslaunch baxter_gazebo baxter_world.launch
```
and in a new terminal, I run the code:

```
 ./baxter.sh sim
 rosrun baxter_examples joint_position_waypoints.py -l right
```

Unfortunately, the error still occurs.
Do you know any other solutions, thank you very much for your answer and best regards.

sxwgit, 2019-03-18:
I reinstalled the  related librarys and found that the problem was solved.
Although I still don't know the cause of the error, it is still very happy to solve it successfully.：）！

yipliu, 2020-11-03:
@sxwgit  I want to know how to reinstall?

gtatiya, 2020-12-14:
I'm also facing same issue. I'm following these steps: https://sdk.rethinkrobotics.com/wiki/Simulator_Installation

These commands:
```
roslaunch baxter_gazebo baxter_world.launch
rosrun baxter_examples joint_velocity_wobbler.py
```

Gives this error:
```
[baxter - http://localhost:11311] baxter@baxter:~/baxter_ws$ rosrun baxter_examples joint_position_keyboard.py
Initializing node... 
Getting robot state... 
Enabling robot... 
[INFO] [WallTime: 1607967394.939458] [31.133000] Robot Enabled
Traceback (most recent call last):
  File "/home/baxter/baxter_ws/src/baxter_examples/scripts/joint_position_keyboard.py", line 157, in <module>
    main()
  File "/home/baxter/baxter_ws/src/baxter_examples/scripts/joint_position_keyboard.py", line 152, in main
    map_keyboard()
  File "/home/baxter/baxter_ws/src/baxter_examples/scripts/joint_position_keyboard.py", line 44, in map_keyboard
    left = baxter_interface.Limb('left')
  File "/home/baxter/baxter_ws/src/baxter_interface/src/baxter_interface/limb.py", line 121, in __init__
    timeout_msg=err_msg)
  File "/home/baxter/baxter_ws/src/baxter_interface/src/baxter_dataflow/wait_for.py", line 55, in wait_for
    raise OSError(errno.ETIMEDOUT, timeout_msg)
OSError: [Errno 110] Left limb init failed to get current joint_states from robot/joint_states

Exiting example...
Disabling robot...
[INFO] [WallTime: 1607967396.236390] [32.166000] Robot Disabled
```
I'm using Ubuntu 14.04 and ROS indigo.

Please help!

yipliu, 2020-12-15:
> I'm also facing same issue. I'm following these steps: https://sdk.rethinkrobotics.com/wiki/Simulator_Installation
>
> These commands:
>
> ```
> roslaunch baxter_gazebo baxter_world.launch
> rosrun [...]

sometime just restart your PC can resolve your problem

gtatiya, 2020-12-15:
> sometime just restart your PC can resolve your problem

I tried restarting the PC, but its still not working, please help!

rojas70, 2021-04-16:
I used to run Baxter very fast and without problems in Kinetic. But since I updated to noetic, I consistently get a slowdown; including using the real robot. I cannot figure out what I time out so much.I have updated my repo's to reflect python3 updates in print, exceptions, range, imports, etc... 

Any generic clues about the slowdown due to a switch in noetic are appreciated.

Rawbotic, 2023-04-28:
noetic,kind of same problem!
[image]

Rawbotic, 2023-04-28:
real robot

Rawbotic, 2023-04-28:
anyone help help would be so helpful

ShawnZhou997, 2024-11-19:
Hi, all. I've solved this by recorrecting my _baxter.sh_ file in the line of `you_ip="192.168.x.xxx"` or `your_hostname="xxxx"`.
This happens after my Internet changed.
Hope this works for you.