RethinkRobotics/baxter_simulator #125: Building error on Kinetic: protoc version

7675t, 2018-06-19:
I have followed this instruction (on the ROS Kinetic tabs): http://sdk.rethinkrobotics.com/wiki/Simulator_Installation

I got errors on protobuf version:
```
In file included from /usr/include/gazebo-7/gazebo/msgs/MessageTypes.hh:7:0,
                 from /usr/include/gazebo-7/gazebo/msgs/msgs.hh:32,
                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /opt/ros/kinetic/include/gazebo_ros_control/gazebo_ros_control_plugin.h:52,
                 from /home/tajima/Codes/tmp_ws/src/baxter_simulator/baxter_gazebo/src/baxter_gazebo_ros_control_plugin.cpp:42:
/usr/include/gazebo-7/gazebo/msgs/altimeter.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
```
The environment is ROS Kinetic, Ubuntu16.04, all packages(including protobuf-compiler, version is  3.1.0) are upgraded. How can I build the baxter simulation? Thanks.

k-okada, 2018-06-19:
did you use https://github.com/RethinkRobotics/baxter_simulator/blob/kinetic-devel/baxter_simulator.rosinstall workspace?

7675t, 2018-06-19:
Yes, I followed the official instruction and it is identical to you've shown.

k-okada, 2018-06-19:
how did you install protobuf-compile 3 ?

https://packages.ubuntu.com/xenial/protobuf-compiler
```
k-okada@p51s:~$ apt show protobuf-compiler
Package: protobuf-compiler
Version: 2.6.1-1.3
Priority: extra
```

7675t, 2018-06-19:
Uh, I see. I completely forget I had installed ppa version of protobuf for some other package (may be google cartographer). 
I did ppa-purge and now I can build all. Thanks, and sorry for bothering.

closed 2018-06-19 by 7675t