RethinkRobotics/baxter_simulator #136: Found unsuitable Qt version "5.12.8" from /usr/bin/qmake, this code   requires Qt 4.x

ashfaq92, 2023-06-05:
I have been trying to install baxter simulator by following the official [link](https://sdk.rethinkrobotics.com/wiki/Simulator_Installation). However, I am stuck in the **Build Source** section where `catkin_make` give following error:

```
CMake Error at /usr/share/cmake-3.16/Modules/FindQt4.cmake:1314 (message):
  Found unsuitable Qt version "5.12.8" from /usr/bin/qmake, this code
  requires Qt 4.x
Call Stack (most recent call first):
  baxter_simulator/baxter_sim_io/CMakeLists.txt:15 (find_package)
```
Here are my implementation details:
```
ROS Version: noetic (I figured everything else for neotic)
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal
```

LiferLifer, 2023-12-31:
I have the same problem, try below:

Install Qt4:
https://ubuntuhandbook.org/index.php/tag/install-qt4-ubuntu-20-04/

(There are other methods to install, but many of resource are no longer valid. If you want, you can try build from the source: 
https://download.qt.io/archive/qt/ )

The key point is to install **qmake** of qt4.

Then, view the current default version:

```shell
qmake -v
```

Open config:

```shell
cd /usr/lib/x86_64-linux-gnu/qt-default/qtchooser
sudo gedit default.conf
```

In it, edit "qt4" to "qt5".

And then, view the current default version again.

```shell
qmake -v
```

May.. can help(

knorth55, 2024-01-03:
Please try this PR.
https://github.com/RethinkRobotics/baxter_simulator/pull/132